thijse/Arduino-EEPROMEx #7: Doesn't write properly sometimes

Cysign, 2015-09-13:
Hey guys,
I'm using this library in a project and figured out that it doesn't always save data to EEPROM properly.
In general this library works great, but sometimes there's no write/update mady where it should.

In my project I'm using the following librarys:
#include "sdelay.h"
#include <Wire.h> 
#include <OneWire.h>
#include <DallasTemperature.h>
#include <LiquidCrystal_I2C.h>
#include <EEPROMex.h>
#include <Time.h>
#include <DS1307RTC.h>

How can I figure out which version of EEPROMex I'm using?
My EEPROMex-folder is named "EEPROMEx-9.1"...but I can't remember if this was the default folder name or I changed it.

per1234, 2015-09-13:
The problem is that the author has the library in debug mode by default so the max number of writes is 100. If you still need debug output you can change the _allowedWrites value using setMaxAllowedWrites() or if you don't need debug output you should comment out line 30 of EEPROMex.cpp to reduce memory usage and make EEPROMex run faster. I submitted https://github.com/thijse/Arduino-EEPROMEx/pull/3 to disable debug mode by default a couple months ago.

Cysign, 2015-09-13:
Is max. No of writes once per uploaded sketch or once the circuit is powered?
That might fix my issues. I'll check this, thanks!

per1234, 2015-09-13:
Per each time the circuit is powered on or the MCU is reset. Note this is the total number of EEPROM writes, not the number of writes to an EEPROM address.

Cysign, 2015-09-13:
Any chance I could disable debug in my code instead ot the EEPROMex.cpp file?

per1234, 2015-09-13:
No way I know of to do this with the way the library is currently written. The new stock Arduino EEPROM library actually does almost everything EEPROMex does except for getAddress() and I wrote [a library](https://github.com/per1234/EEPROMallocation) that does that separately.

Cysign, 2015-09-13:
Thanks for your support!

closed 2015-09-13 by Cysign