thijse/Arduino-EEPROMEx #9: Not able to Locate EEPROM.h file

shirish47, 2015-12-17:
C:\Program Files (x86)\Arduino\libraries\EEPROMex/EEPROMex.h:23:20: fatal error: EEPROM.h: No such file or directory
 #include <EEPROM.h>
                    ^
compilation terminated.
Error compiling.
 
If I include EEPROM in main Code I get this problem.. I am using Arduino 1.6.3

C:\Program Files (x86)\Arduino\libraries\EEPROMex/EEPROMVar.h:1:22: fatal error: WProgram.h: No such file or directory
 #include <WProgram.h>
                      ^
compilation terminated.
Error compiling.

shirish47, 2015-12-17:
libraries\EEPROMex/EEPROMex.h:159:22: error: conflicting declaration 'EEPROMClassEx EEPROM'
 extern EEPROMClassEx EEPROM;
                      ^
In file included from EEPROM.ino:1:0:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM/EEPROM.h:145:20: error: 'EEPROM' has a previous declaration as 'EEPROMClass EEPROM'
 static EEPROMClass EEPROM;
                    ^

DoctorTardi, 2016-01-21:
I am having exactly the same problem, did you manage to resolve it?

shirish47, 2016-01-27:
I did not use that library later.. :P :D

thijse, 2016-06-02:
I think the problem is that both EEPROMex.h & EEPROM.h are being included (may be indirectly) and both libraries expose an extern EEPROM object.

Can you send me the sketch you are using and the board?

thijse, 2016-06-02:
well, not send me the actual board, but the type of the board :)

shirish47, 2016-06-03:
well that board was custom built for a project. but controller was Atmega328 same are everything was same as arduino mini pro. now I don't remember the file . :P but thanks in case if I again fall into this trouble I would post it.