thijse/Arduino-EEPROMEx #34: ESP8266 avr/eeprom.h library issue

Luke8900, 2020-09-03:
I was wondering if it was possible to solve this issue of using the EEPROMex library with an ESP8266 device (such as node MCU) when I try to compile the code I get an error saying Avr/EEPROM.h no such file in the directory and I did a bit of research which said that this library cannot be used with ESP32 devices (I'm not sure if that is the same for ESP8266 I'm assuming it is) so if there is a fix or a change I could make to make the code work for my node MCU it would be a great help, thanks.

hervema, 2020-12-14:
Hello, Based on the previous comment, could you please confirm that your library CAN NOT be used on ESP8266 ou ESP32 platform ? 

Thanks in advance, 

Hervema

SniffyMcTasty, 2023-02-23:
Yes, I have the same problem, the library cannot be used with ESP32.